davesteele/comitup #208: Quiestion - access to internet in AP mode

Ataraxiall, 2022-05-30:
Hi!

It´s possible have internet access if the ethernet cable is connected. In some cases, i want to access to internet.

Thanks!

davesteele, 2022-05-31:
If the Ethernet connection has an upstream connection and is using DHCP, then it will configure to use that when connected.

davesteele, 2022-05-31:
Note that this capability is part of the Comitup image, not the package itself.

Ataraxiall, 2022-06-29:
Thanks for the reply.

I have already been able to make it work, the only thing I added is this command
"iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE" and "echo 1 > /proc/sys/net/ipv4/ip_forward
"The "problem" is that from another connected computer I can put a manual dns on it and the internet works, but from the mobile it does not browse, it always takes me to the comitup website for wifi selection. Can it be changed so that comitup assigns a dns different from the original (10.41.0.1)?

Tranks!

davesteele, 2022-06-30:
Upon reflection, there are multiple strategies for implementing the captive portal - both defined in dns-hotspot.conf. Option 160 explicitly defines a captive portal url. Option 6 defines a dns server to be used, which, in this case, returns the IP of the device for all requests.

You MAY be able to switch option 6 to a real dns server and still have a working captive portal, relying on option 160. I have no idea which platforms would break from this.

If it does work "correctly", you probably won't like it. It will still show the portal pages for all web requests (in HOTSPOT mode), while routing all other traffic over the external connection.

It's been several years since I worked through this, and, at the time, determined that I needed both.

Bottom line, #168.