davesteele/comitup #209: Question - Ubuntu 20.04/22.04 support

JohannesArtmeier, 2022-06-09:
Hey,

does this work with newer versions of ubuntu which changed to netplan? Is there anything special to consider?

davesteele, 2022-06-09:
I haven't tried it, and haven't used netplan, but, according to [this](https://ubuntu.com/core/docs/networkmanager/networkmanager-and-netplan), it should work.

NetworkManager needs to be in control of networking.

JohannesArtmeier, 2022-06-09:
I tried a bit around with the latest ubuntu 22.04, but it does not work. Any chance there will be an update to netplan?

Thanks!

davesteele, 2022-06-09:
I will not address this anytime soon.

Though, as always, I promise to promptly review patches.

Naterd, 2022-07-03:
I was able to get it working on 22.04 except for ssid scanning
```## Install Ubuntu 22.04 lts
## AS ROOT
systemctl disable systemd-networkd.service
systemctl mask systemd-networkd.service
systemctl stop systemd-networkd.service
apt-get install network-manager -y
apt install vim -y
netplan generate
systemctl unmask NetworkManager
systemctl enable NetworkManager
systemctl start NetworkManager
apt-get install --reinstall resolvconf -y
systemctl stop systemd-resolved
systemctl mask systemd-resolved
sytemctl disable systemd-resolved
vim /etc/resolv.conf and set nameserver to 1.1.1.1
resolvconf -u # this disables updating and fixes resetting problem
apt install comitup -y
## There is a bad bug in python-networkmanager that causes comitup to crash, once this is merged in and ubuntu picks it up this may no longer be needed but do it for now https://github.com/seveas/python-networkmanager/issues/93
wget https://davesteele.github.io/comitup/deb/python3-networkmanager_2.2-1dss01_all.deb
dpkg -i python3-networkmanager_2.2-1dss01_all.deb
## Some of these services may not exist but it doesnt hurt to make sure they never conflict with comitup
rm /etc/network/interfaces
systemctl mask dhcpd.service
systemctl mask wpa-supplicant.service
systemctl restart comitup
```

I keep seeing 

```Jul 03 15:33:05 look-test comitup[6101]: command failed: Operation not supported (-95)
Jul 03 15:33:06 look-test comitup[6111]: command failed: Operation not supported (-95)
Jul 03 15:33:06 look-test comitup[6124]: command failed: Operation not supported (-95)
Jul 03 15:33:06 look-test comitup[6134]: command failed: Operation not supported (-95)
Jul 03 15:33:14 look-test comitup[6150]: command failed: Operation not supported (-95)
Jul 03 15:33:22 look-test comitup[6165]: command failed: Operation not supported (-95)
```

In the service logs, manually entering ssid works. Any idea why this is happening?

I have confirmed my device is able to scan for ssid by running `iw dev <device> scan` .

Naterd, 2022-07-03:
```Jul 03 16:05:04 look-test NetworkManager[7741]: <debug> [1656885904.4117] device[6950fb7ed3312086] (wlp0s20f3): wifi-ap: dump    FC:EC:DA:15:59:2C "PorKayFi"                          [ a 161  29% P_ __ W:0000 R:0188 ]  62.823s sup:/7 [nm:/8]
Jul 03 16:05:04 look-test NetworkManager[7741]: <debug> [1656885904.4117] device[6950fb7ed3312086] (wlp0s20f3): wifi-ap: dump    CA:2F:F7:1B:0C:8A (none)                              [ a  11  32% P_ __ W:0000 R:0188 ]  64.822s sup:/8 [nm:/9]
Jul 03 16:05:04 look-test NetworkManager[7741]: <debug> [1656885904.4118] device[6950fb7ed3312086] (wlp0s20f3): wifi-ap: dump    0A:2F:F7:1B:0C:8A (none)                              [ a  11  32% P_ __ W:0000 R:0188 ]  64.822s sup:/9 [nm:/10]
Jul 03 16:05:04 look-test NetworkManager[7741]: <debug> [1656885904.4118] device[6950fb7ed3312086] (wlp0s20f3): wifi-ap: dump    78:45:58:F5:CA:55 "PorKayFi"                          [ a  48  25% P_ __ W:0000 R:0188 ]  64.822s sup:/10 [nm:/11]
Jul 03 16:05:04 look-test NetworkManager[7741]: <debug> [1656885904.4118] device[6950fb7ed3312086] (wlp0s20f3): wifi-ap: dump    DA:2F:F7:1B:0C:8A "BB11"                              [ a  11  30% 
```
I can see networkmanager scanning SSIDs when I connect to comitup hotspot

davesteele, 2022-07-03:
I would expect a Traceback in one of the logs from that error. That output would be useful.

Or, stop the service and run "comitup" as root from the command line, to see error infomation on the console.

It may help to change INFO to DEBUG in /usr/share/comitup/comitup/comitup.py.

The output you posted has a reference to https://github.com/seveas/python-networkmanager/issues/93. That issue can be a problem for comitup (the app crashes if one of the connections doesn't have DNS defined). I added a [patched NetworkManager](https://davesteele.github.io/comitup/latest/python3-networkmanager_latest.html) to the Comitup repository to correct the problem.

Naterd, 2022-07-04:
```2022-07-03 20:25:21,338 - comitup - DEBUG - Calling GetSettings
2022-07-03 20:25:21,338 - comitup - DEBUG - Calling nm.ActivateConnection()
2022-07-03 20:25:21,345 - comitup - DEBUG - stopping comitup-web.service web service
2022-07-03 20:25:21,353 - comitup - DEBUG - Running dnsmasq using
2022-07-03 20:25:21,360 - comitup - DEBUG - Kicking wpa on wlp0s20f3
2022-07-03 20:25:21,368 - comitup - DEBUG - nmm - primary state 110
2022-07-03 20:25:21,368 - comitup - DEBUG - nmm - secondary state 110
2022-07-03 20:25:21,397 - comitup - DEBUG - nmm - primary state 30
2022-07-03 20:25:21,398 - comitup - DEBUG - nmm - secondary state 30
2022-07-03 20:25:21,407 - comitup - DEBUG - nmm - primary state 40
2022-07-03 20:25:21,407 - comitup - DEBUG - nmm - secondary state 40
2022-07-03 20:25:21,413 - comitup - DEBUG - nmm - primary state 50
2022-07-03 20:25:21,414 - comitup - DEBUG - nmm - secondary state 50
2022-07-03 20:25:21,415 - comitup - DEBUG - nmm - primary state 60
2022-07-03 20:25:21,415 - comitup - DEBUG - nmm - secondary state 60
2022-07-03 20:25:21,417 - comitup - DEBUG - nmm - primary state 40
2022-07-03 20:25:21,417 - comitup - DEBUG - nmm - secondary state 40
2022-07-03 20:25:21,420 - comitup - DEBUG - nmm - primary state 50
2022-07-03 20:25:21,420 - comitup - DEBUG - nmm - secondary state 50
2022-07-03 20:25:45,962 - comitup - DEBUG - Getting device list from 'iw'
2022-07-03 20:25:45,968 - comitup - DEBUG - Getting AP list from 'iw' dev wlp0s20f3
2022-07-03 20:25:47,851 - comitup - DEBUG - Got an exception, returning None, get_active_ip
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nm.py", line 60, in wrapper
    return fp(*args, **kwargs)
  File "/usr/share/comitup/comitup/nm.py", line 142, in get_active_ip
    return device.Ip4Config.Addresses[0][0]
IndexError: list index out of range
2022-07-03 20:25:47,857 - comitup - DEBUG - Got an exception, returning None, get_active_ip
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nm.py", line 60, in wrapper
    return fp(*args, **kwargs)
  File "/usr/share/comitup/comitup/nm.py", line 142, in get_active_ip
    return device.Ip4Config.Addresses[0][0]
AttributeError: 'NoneType' object has no attribute 'Addresses'
2022-07-03 20:25:47,893 - comitup - DEBUG - Got an exception, returning None, get_active_ip
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nm.py", line 60, in wrapper
    return fp(*args, **kwargs)
  File "/usr/share/comitup/comitup/nm.py", line 142, in get_active_ip
    return device.Ip4Config.Addresses[0][0]
IndexError: list index out of range
2022-07-03 20:25:47,899 - comitup - DEBUG - Got an exception, returning None, get_active_ip
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nm.py", line 60, in wrapper
    return fp(*args, **kwargs)
  File "/usr/share/comitup/comitup/nm.py", line 142, in get_active_ip
    return device.Ip4Config.Addresses[0][0]
AttributeError: 'NoneType' object has no attribute 'Addresses'
2022-07-03 20:25:47,935 - comitup - DEBUG - Got an exception, returning None, get_active_ip
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nm.py", line 60, in wrapper
    return fp(*args, **kwargs)
  File "/usr/share/comitup/comitup/nm.py", line 142, in get_active_ip
    return device.Ip4Config.Addresses[0][0]
IndexError: list index out of range
2022-07-03 20:25:47,941 - comitup - DEBUG - Got an exception, returning None, get_active_ip
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nm.py", line 60, in wrapper
    return fp(*args, **kwargs)
  File "/usr/share/comitup/comitup/nm.py", line 142, in get_active_ip
    return device.Ip4Config.Addresses[0][0]
AttributeError: 'NoneType' object has no attribute 'Addresses
```

Seeing this after enabling debug. Thank you for looking into this. I also am already using your network manager patched package as well.

Naterd, 2022-07-04:
```022-07-03 20:42:19,974 - comitup - DEBUG - starting comitup-web.service web service
2022-07-03 20:42:19,977 - comitup - DEBUG - Running dnsmasq using /usr/share/comitup/dns/dns-hotspot.conf
2022-07-03 20:42:19,980 - comitup - DEBUG - starting comitup-web.service web service
2022-07-03 20:42:19,983 - comitup - DEBUG - Running dnsmasq using /usr/share/comitup/dns/dns-hotspot.conf
2022-07-03 20:42:20,089 - comitup - DEBUG - starting comitup-web.service web service
2022-07-03 20:42:20,093 - comitup - DEBUG - Running dnsmasq using /usr/share/comitup/dns/dns-hotspot.conf
2022-07-03 20:42:20,201 - comitup - DEBUG - starting comitup-web.service web service
2022-07-03 20:42:20,206 - comitup - DEBUG - Running dnsmasq using /usr/share/comitup/dns/dns-hotspot.conf
2022-07-03 20:42:20,320 - comitup - DEBUG - Getting device list from 'iw'
2022-07-03 20:42:20,337 - comitup - DEBUG - Getting AP list from 'iw' dev wlp0s20f3
2022-07-03 20:42:27,897 - comitup - DEBUG - Getting device list from 'iw'
2022-07-03 20:42:27,917 - comitup - DEBUG - Getting AP list from 'iw' dev wlp0s20f3
2022-07-03 20:43:05,561 - comitup - DEBUG - Getting device list from 'iw'
2022-07-03 20:43:05,579 - comitup - DEBUG - Getting AP list from 'iw' dev wlp0s20f3
2022-07-03 20:43:06,156 - comitup - DEBUG - Getting device list from 'iw'
2022-07-03 20:43:06,176 - comitup - DEBUG - Getting AP list from 'iw' dev wlp0s20f3
2022-07-03 20:43:06,336 - comitup - DEBUG - Getting device list from 'iw'
2022-07-03 20:43:06,353 - comitup - DEBUG - Getting AP list from 'iw' dev wlp0s20f3
2022-07-03 20:43:06,614 - comitup - DEBUG - Getting device list from 'iw'
2022-07-03 20:43:06,633 - comitup - DEBUG - Getting AP list from 'iw' dev wlp0s20f3
```

Just seeing this over and over when connecting to captive portal, manually entering SSID works

Naterd, 2022-07-04:
Running iwscan directly outputs command not supported
```root@look-test:~# python3 /usr/share/comitup/comitup/iwscan.py
command failed: Operation not supported (-95)
{}
[]
0
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/iwscan.py", line 146, in <module>
    for ap in devaps(sys.argv[1], sys.stdin.read()):
IndexError: list index out of range
```

Naterd, 2022-07-04:
We can see 0 points are being passed to web service
```root@look-test:/var/log# tail -F comitup-web.log
2022-07-04 14:50:05,733 - comitup_web - INFO - Starting comitup-web
2022-07-04 14:53:41,410 - comitup_web - INFO - Starting comitup-web
2022-07-04 14:57:09,151 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:00:54,271 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:04:28,875 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:08:04,700 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:11:16,349 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:11:54,931 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:13:28,977 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:17:01,129 - comitup_web - INFO - Starting comitup-web
2022-07-04 15:17:34,498 - comitup_web - INFO - index.html - 0 points
2022-07-04 15:17:35,351 - comitup_web - INFO - index.html - 0 points
2022-07-04 15:17:35,440 - comitup_web - INFO - index.html - 0 points
2022-07-04 15:17:35,722 - comitup_web - INFO - index.html - 0 points
2022-07-04 15:17:58,400 - comitup_web - INFO - confirm.html - ssid cagetopia, mode single
2022-07-04 15:17:58,490 - comitup_web - INFO - index.html - 0 points
2022-07-04 15:18:06,349 - comitup_web - INFO - connect.html - ssid cagetopia
2022-07-04 15:18:06,473 - comitup_web - INFO - index.html - 0 points
2022-07-04 15:18:07,352 - comitup_web - DEBUG - Calling client connect
```

davesteele, 2022-07-05:
Your logs suggest you may be running a pretty old version of Comitup - the one included in Ubuntu 22.04. Current is 1.37-1. Consider the [repo](https://davesteele.github.io/comitup/ppa.html).

Note that I incorrectly referred to python3-networkmanager as NetworkManager in my comment above.

Try "sudo python3 iwscan.py wlan0" (^d to terminate the stdin read).

Naterd, 2022-07-05:
```root@look-test:~# apt list comitup
Listing... Done
comitup/stable,stable,now 1.37-1 all [installed]
```
Now running on updated version as well as version 2 of patched python-networkmanager.

 Behavior appears to be the same

```Jul 04 19:53:00 look-test dnsmasq[331089]: warning: interface wlp0s20f3 does not currently exist
Jul 04 19:53:00 look-test dnsmasq-dhcp[331089]: DHCP, IP range 10.41.0.100 -- 10.41.0.150, lease time 10m
Jul 04 19:53:00 look-test dnsmasq[331089]: reading /etc/resolv.conf
Jul 04 19:53:00 look-test dnsmasq[331089]: using nameserver 1.1.1.1#53
Jul 04 19:53:00 look-test dnsmasq[331089]: read /etc/hosts - 7 addresses
Jul 04 19:53:00 look-test dnsmasq[331089]: exiting on receipt of SIGTERM
Jul 04 19:53:28 look-test dnsmasq[331155]: started, version 2.86 cachesize 150
Jul 04 19:53:28 look-test dnsmasq[331155]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile
Jul 04 19:53:28 look-test dnsmasq-dhcp[331155]: DHCP, IP range 10.41.0.100 -- 10.41.0.150, lease time 10m
Jul 04 19:53:28 look-test dnsmasq[331155]: reading /etc/resolv.conf
Jul 04 19:53:28 look-test dnsmasq[331155]: using nameserver 1.1.1.1#53
Jul 04 19:53:28 look-test dnsmasq[331155]: read /etc/hosts - 7 addresses
Jul 04 19:53:29 look-test comitup[331162]: command failed: Operation not supported (-95)
Jul 04 19:53:45 look-test dnsmasq-dhcp[331155]: DHCPDISCOVER(wlp0s20f3) b6:5f:8f:b7:44:fc
Jul 04 19:53:45 look-test dnsmasq-dhcp[331155]: DHCPOFFER(wlp0s20f3) 10.41.0.100 b6:5f:8f:b7:44:fc
Jul 04 19:53:46 look-test dnsmasq-dhcp[331155]: DHCPREQUEST(wlp0s20f3) 10.41.0.100 b6:5f:8f:b7:44:fc
Jul 04 19:53:46 look-test dnsmasq-dhcp[331155]: DHCPACK(wlp0s20f3) 10.41.0.100 b6:5f:8f:b7:44:fc
Jul 04 19:53:48 look-test comitup[331175]: command failed: Operation not supported (-95)
Jul 04 19:54:15 look-test comitup[331195]: command failed: Operation not supported (-95)
Jul 04 19:54:25 look-test comitup[331203]: command failed: Operation not supported (-95)
Jul 04 19:54:32 look-test comitup[331212]: command failed: Operation not supported (-95)
Jul 04 19:54:40 look-test comitup[331220]: command failed: Operation not supported (-95)
Jul 04 19:54:44 look-test comitup[331228]: command failed: Operation not supported (-95)
Jul 04 19:54:45 look-test comitup[331236]: command failed: Operation not supported (-95)
Jul 04 19:55:32 look-test dnsmasq-dhcp[331155]: DHCPDISCOVER(wlp0s20f3) b6:5f:8f:b7:44:fc
Jul 04 19:55:32 look-test dnsmasq-dhcp[331155]: DHCPOFFER(wlp0s20f3) 10.41.0.100 b6:5f:8f:b7:44:fc
Jul 04 19:55:33 look-test dnsmasq-dhcp[331155]: DHCPREQUEST(wlp0s20f3) 10.41.0.100 b6:5f:8f:b7:44:fc
Jul 04 19:55:33 look-test dnsmasq-dhcp[331155]: DHCPACK(wlp0s20f3) 10.41.0.100 b6:5f:8f:b7:44:fc
Jul 04 19:55:35 look-test comitup[331250]: command failed: Operation not supported (-95)
Jul 04 19:55:41 look-test comitup[331275]: command failed: Operation not supported (-95)
Jul 04 19:55:51 look-test comitup[331285]: command failed: Operation not supported (-95)
```

when comitup is not in hotspot mode
```root@look-test:~# python3 /usr/share/comitup/comitup/iwscan.py wlan0
[{'ssid': 'cagetopia', 'strength': '68.0', 'security': 'encrypted'}, {'ssid': 'Cook_Gear_Kids', 'strength': '64.0', 'security': 'encrypted'}, {'ssid': 'Cook_Zyxel', 'strength': '52.0', 'security': 'encrypted'}, {'ssid': 'NETGEAR45', 'strength': '46.0', 'security': 'encrypted'}, {'ssid': 'PorKayFi', 'strength': '44.0', 'security': 'encrypted'}, {'ssid': 'Wolfie2.4', 'strength': '42.0', 'security': 'encrypted'}, {'ssid': 'TheBest', 'strength': '32.0', 'security': 'encrypted'}, {'ssid': 'PorKay Guest', 'strength': '28.0', 'security': 'encrypted'}, {'ssid': 'Black Guest', 'strength': '22.0', 'security': 'encrypted'}, {'ssid': 'HT', 'strength': '22.0', 'security': 'encrypted'}, {'ssid': 'Black', 'strength': '20.0', 'security': 'encrypted'}]
1
```

hotspot mode:
```root@look-test:~# python3 /usr/share/comitup/comitup/iwscan.py wlan0
command failed: Operation not supported (-95)
[]
0
```

I am not sure if this is intended behavior

Any idea why the data is dropping when being passed to the web service? Manually entering SSID works on latest version.

davesteele, 2022-07-05:
If the service crashes/restarts, the comitup-cli connection is lost, and has to be restarted.

It's not intended behavior, but I don't know what is going on right now.

Naterd, 2022-07-10:
@davesteele Any insights on where I should look? Or any ideas etc? I am willing to try to possibly fix this, I just figure anything that pops into your head could potentially save me a lot of time.

davesteele, 2022-07-10:
Start by finding any tracebacks in e.g. daemon.log, or run in the terminal, with possibly debug-level logs.

Naterd, 2022-07-28:
The wifi driver for the Intel NUC does not support scanning for access points while already in AP mode.

Comitup logic is to scan for access points while also being an access point itself so this would not work at all for my wifi driver. [https://github.com/davesteele/comitup/blob/main/comitup/statemgr.py#L71](url)

I was able to work around this problem by modifying statemgr to scan for APs at startup and storing that in a global variable before it entered Access point mode, thus populating the SSIDs correctly when loading the web portal.

The drawback to this is it will not continue to scan for new SSIDs, you may need to restart the service to pick up new SSIDS if they were set up after the commitup service had started.

davesteele, 2022-07-28:
... or add another USB WiFi adapter. Comitup will make use of it to do the scan.

Be sure to set [*enable_appliance_mode*](https://davesteele.github.io/comitup/man/comitup-conf.pdf) according to your wishes.

Naterd, 2022-07-28:
> 

Yes unfortunately that was not an option for me

docgalaxyblock, 2022-08-09:
Greetings I tried to get comitup working for the [OpenMower project](https://github.com/ClemensElflein/OpenMower)
 and finally got it working without stopping/disabling/masking `systemd-resolved` on `ubuntu-20.04.4-preinstalled-server-arm64+raspi`

I found a hint at the [pi-hole install instructions](https://github.com/pi-hole/docker-pi-hole/#installing-on-ubuntu)

After this small hint I got it working like that:
```
wget https://davesteele.github.io/comitup/deb/python3-networkmanager_2.2-1dss02_all.deb
sudo dpkg -i python3-networkmanager_2.2-1dss02_all.deb
rm python3-networkmanager_2.2-1dss02_all.deb
sudo apt install -f
# we need to disable systemd-resolveds caching DNS stub resolver
# if you want to know why check: https://github.com/pi-hole/docker-pi-hole/#installing-on-ubuntu
sudo sed -r -i.orig 's/#?DNSStubListener=yes/DNSStubListener=no/g' /etc/systemd/resolved.conf
sudo systemctl restart systemd-resolved.service

nslookup ntp.org
# this will throw an dns error:
#   ;; connection timed out; no servers could be reached
sudo sh -c 'rm /etc/resolv.conf && ln -s /run/systemd/resolve/resolv.conf /etc/resolv.conf'
sudo systemctl restart systemd-resolved.service
nslookup ntp.org
# DNS should work now

# Install comitup using latest version from the website
wget https://davesteele.github.io/comitup/deb/comitup_1.38-1_all.deb
sudo dpkg -i comitup_1.38-1_all.deb
rm comitup_1.38-1_all.deb
sudo apt install -f
sudo systemctl stop comitup
sudo nmtui
sudo reboot
# it should work now
```

🤗

docgalaxyblock, 2022-08-12:
Now I tried to install it following the github wiki guide and 
```
wget https://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.deb
sudo dpkg -i --force-all davesteele-comitup-apt-source_latest.deb
rm davesteele-comitup-apt-source_latest.deb
sudo apt update
```
the apt sources.d listing:
```
deb [signed-by=/usr/share/keyrings/davesteele-archive-keyring.gpg] http://davesteele.github.io/comitup/repo comitup main
```
and apt mentions:
```
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://davesteele.github.io/comitup/repo comitup InRelease' doesn't support architecture 'arm64'
```
@davesteele do you an idea what is wrong here?

davesteele, 2022-08-12:
I had not included arm64 in the repository.

Try it again, and tell me how it works.

docgalaxyblock, 2022-08-12:
@davesteele 👍 It works now - Thanks

Did you also read my previous post regarding `systemd-resolved`?
Would that be an adequate solution to completely disable resolved?

davesteele, 2022-08-12:
I haven't looked into that yet.

davesteele, 2022-09-21:
I added a [link](https://github.com/davesteele/comitup/wiki/Installing-Comitup/_compare/ab193a80ba47b1d52f2a5fd8fe38a45a6bbf3859...4756724e4fbc2d3742e3989783416c2007951086) in the [Installing](https://github.com/davesteele/comitup/wiki/Installing-Comitup) Wiki page to this method.