davesteele/comitup #212: Web service won't start DBus exception

wspi, 2022-07-05:
I've just downloaded the [comitup image](https://steele.debian.net/comitup/image_2022-04-16-Comitup.zip) and put on a raspberry pi 3 B but the web page won't start.. I've tried to run the binary `comitup-web` and got this error:

```
ERROR:dbus.proxies:Introspect error on :1.20:/com/github/davesteele/comitup: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
	Traceback (most recent call last):
  File "/usr/sbin/comitup-web", line 33, in <module>
    sys.exit(load_entry_point('comitup==1.34', 'console_scripts', 'comitup-web')())
  File "/usr/share/comitup/web/comitupweb.py", line 173, in main
    ciu_client.ciu_state()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

I've found some closed issues asking to upgrade [python network manager](//davesteele.github.io/comitup/deb/python3-networkmanager_2.2-1dss02_all.deb) but didn't solve

not sure if there is any other workaround or some latest image with this fix

davesteele, 2022-07-05:
Note that an apt upgrade will get you a version that eliminates service crashes when APs containing extended unicode characters are encountered. That crash would show up as a Dbus noreply.

wspi, 2022-07-06:
I've just did an apt upgrade and the issue persists

the package was upgraded to 1.37-1

```
ii  comitup                              1.37-1                           all          bootstrap Wi-Fi using Wi-Fi
ii  comitup-watch                        0.2-1                            all          monitor local Comitup-enabled devices
ii  davesteele-comitup-apt-source        1.2                              all          Apt configuration for comitup repository
```

```
ERROR:dbus.proxies:Introspect error on :1.24:/com/github/davesteele/comitup: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Traceback (most recent call last):
  File "/usr/sbin/comitup-web", line 33, in <module>
    sys.exit(load_entry_point('comitup==1.37', 'console_scripts', 'comitup-web')())
  File "/usr/share/comitup/web/comitupweb.py", line 173, in main
    ciu_client.ciu_state()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

wspi, 2022-07-06:
I was scanning the SSIDs looking for weird chars and found one that has `'` in it's name, not sure if that could be the reason

davesteele, 2022-07-06:
#211

wspi, 2022-07-06:
@davesteele I've taken the pi to an underground garage and it worked, so probably it's the apostrophe or any other char in an SSID, #211 didn't solved, but I'll try to fix it in the code as I can reproduce and let you know, I'm not sure you want to close this issue

closed 2022-07-06 by wspi