davesteele/comitup #213: Question: Best way to execute python code after raspberry pi has connected to router?

gbernal, 2022-07-26:
Dear @davesteele,

Thank you first of all for sharing your work. I'm using comitup for a research project on a raspberry pi 3B and the IP of the device changes every time the pi is connected to user's local router. 

I would like to send to my server the IP address so I can then ssh into the device without having to connect to a monitor. I have seen at StackOverflow that people suggest putting the python code here `"/etc/network/if-up.d/upstart" just after "all_interfaces_up" ` https://stackoverflow.com/questions/29204423/how-to-automatically-run-python-script-after-raspberry-pi-fully-boots 

But since network interface is handled by comitup I figured that it might be better to ask here. 

Thank you in advance.

davesteele, 2022-07-26:
1. Comitup publishes the ip address on all interfaces. If you have mdns support, just attempt "ssh comitup@&lt;service_name&gt;.local. The *service_name* is the name used for the SSID in HOTSPOT mode.
1. If you place an executable script at */usr/local/bin/comitup-callback*, it will be called with the first argument = "*CONNECTED*" when the upstream connection is successful. See the *external_callback* parameter on the [comitup.conf man page](https://davesteele.github.io/comitup/man/comitup-conf.pdf).
1. Use [comitup-watch](https://github.com/davesteele/comitup-watch) to discover IP addresses of connected Comitup devices.

gbernal, 2022-07-26:
Wonderful! Thank you @davesteele!

gbernal, 2022-07-26:
Dear @davesteele 

One more question, can you please clarify what does comitup-callback mean in this path ?/usr/local/bin/comitup-callback.
Is comitup-callback my script or a directory? My RPi only had usr/local/bin/ but nothing after that. Is that a path I have to create?
[image]

Also, Am I right to assume that I have to modify  comitup.conf from `# external_callback: /usr/local/bin/comitup-callback` to `external_callback: /usr/local/bin/comitup-callback/myscript.py` ?

Thank you again.

davesteele, 2022-07-26:
*comitup-callback* is an executable script that is your responsibility. The path is defined by the *external_callback* parameter, but, if you don't define that parameter, it takes the value shown in the *comitup.conf* comment. So, no, you don't have to make that change.

closed 2022-07-26 by gbernal