davesteele/comitup #214: comitup-cli x

qwksilver, 2022-09-04:
comitup-cli x

version 1.38-1 from apt-get on buster, manual install
buster is 2021-05-07-raspios-buster-armhf
stock from pi image no upgrade (upgrade not doable at this time (it breaks other things for me))
bullseye no doable for same reason

my understanding is that the x command deletes all current connections, and resets the comitup returning it to hotspot state
currently x does nothing visable, and the hotspot returns, sill connected to last connection.

comitup-cli d x 
is what i am using instead, but it likely is only deleting current connection, and not all connections.

if this is correct sorry for the issue entry, if this is not correct, thank you for your help.

davesteele, 2022-09-04:
"*comitup-cli d x*" will delete the current connection. The argument is ignored.

The nuke option requires that *enable_nuke* is enabled in the [configuration file](https://davesteele.github.io/comitup/man/comitup-conf.pdf) (the command is dangerous, and not useful for most).

The connections are managed by NetworkManager. You can view them with "*nmcli con*" and delete one with "*nmcli con del x*".

qwksilver, 2022-09-04:
ah, can you elaborate as to what makes the nuke option dangerous.

a slightly unrelated question too:
is there a preferred deactivation procedure that doesn't take comitup completely out of the unit, but would allow the gui to manage connections?

davesteele, 2022-09-06:
It's just dangerous in that can delete bunches of connections, and is easy to activate from *comitup-cli*.

The preferred deactivation procedure is to stop or disable the *comitup.service* systemd service. Current active connections are preserved, but it won't interfere with any changes you make.

closed 2024-07-10 by davesteele