davesteele/comitup #215: Information, installation script.

qwksilver, 2022-09-05:
Below is a script I developed, from your instructions, that saves the original files if someone were to want to back out.  This also seems to consistently install the current version without issues.  Feel free to adopt or share this file.  The proper file extension is only .sh but github won't let me share it that way.

[comitup.sh.txt](https://github.com/davesteele/comitup/files/9491534/comitup.sh.txt)

davesteele, 2022-09-06:
Looks useful. Note that I haven't tested it.

Anyone coming across this should know that there are [disk images](https://davesteele.github.io/comitup/) of Raspios with comitup installed.

davesteele, 2022-09-06:
I added a link to this page from the [wiki](https://github.com/davesteele/comitup/wiki/Installing-Comitup).

m1dnight, 2022-09-10:
Not sure if this is the right place to start this discussion, but have you had any chance on getting it wokring @qwksilver ? Is there anything I can do to help to get it working on bullseye?

Jeroen6, 2023-03-08:
Hey, thanks. It works. 
Do you think there is a way to make the script not kill the ssh-session?

mpentler, 2023-08-23:
This seems to be missing `systemctl enable NetworkManager.service` which is leading me to reimage yet another time tonight.

davesteele, 2023-08-23:
@mpentler , and all, you need to be aware that there is a [Comitup image](http://davesteele.github.io/comitup/#getting-comitup), which is just Raspberry Pi OS with Comitup installed.

Once you have your reimage done, try running:

    gpg --keyring /usr/share/keyrings/debian-keyring.gpg --no-default-keyring --list-key "David Steele"

The Comitup images are signed with this key.

mpentler, 2023-08-23:
I appreciate that - but lots of us are adding this to something we're already building and thiught "ah - I need to add this functionality!" at a later date. We're not starting from scratch, you know?

Otherwise the script was great, I just wanted to flag it up.

closed 2024-04-10 by davesteele