davesteele/comitup #216: Installation fails now in Raspbian GNU/Linux 11 (bullseye)

outdoorbits, 2022-09-30:
While comitup worked great in Raspbian GNU/Linux 11 (bullseye) until short, now installation breaks.
On a Raspberry Pi 4 without additional wlan-adapters, I followed the installation-steps in [https://github.com/davesteele/comitup/wiki/Installing-Comitup](https://github.com/davesteele/comitup/wiki/Installing-Comitup). 
I can't use your image, because for me comitup has become a very central component of [https://github.com/outdoorbits/little-backup-box](https://github.com/outdoorbits/little-backup-box) and I hope a lot, it will work again soon. ;-) Great project!

Make a note in the issue that you have covered these topics: NOTE

The step
`apt-get install comitup comitup-watch`
delivers this error after all:

`Error: NetworkManager is not running.
Job for comitup.service failed because the control process exited with error code.
See "systemctl status comitup.service" and "journalctl -xe" for details.
comitup-web.service is a disabled or a static unit, not starting it.` 

`sudo systemctl status comitup.service`:
`● comitup.service - Comitup Wifi Management
     Loaded: loaded (/lib/systemd/system/comitup.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2022-09-30 14:58:09 BST; 2s ago
       Docs: man:comitup(8)
    Process: 2020 ExecStart=/usr/sbin/comitup (code=exited, status=1/FAILURE)
   Main PID: 2020 (code=exited, status=1/FAILURE)
        CPU: 335ms`

`journalctl -xe`:
`Sep 30 15:00:48 raspberrypi comitup[2062]: During handling of the above exception, another exception occurred:
Sep 30 15:00:48 raspberrypi comitup[2062]: Traceback (most recent call last):
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/sbin/comitup", line 5, in <module>
Sep 30 15:00:48 raspberrypi comitup[2062]:     from comitup.comitup import main
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/share/comitup/comitup/comitup.py", line 22, in <module>
Sep 30 15:00:48 raspberrypi comitup[2062]:     from comitup import cdns  # noqa
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/share/comitup/comitup/cdns.py", line 14, in <module>
Sep 30 15:00:48 raspberrypi comitup[2062]:     from comitup import modemgr
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/share/comitup/comitup/modemgr.py", line 16, in <module>
Sep 30 15:00:48 raspberrypi comitup[2062]:     from comitup import config, nm
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/share/comitup/comitup/nm.py", line 24, in <module>
Sep 30 15:00:48 raspberrypi comitup[2062]:     import NetworkManager as nm
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 285, in <module>
Sep 30 15:00:48 raspberrypi comitup[2062]:     class NetworkManager(NMDbusInterface):
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 142, in __new__
Sep 30 15:00:48 raspberrypi comitup[2062]:     proxy = init_bus.get_object(type_.dbus_service, attrs['object_path'])
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
Sep 30 15:00:48 raspberrypi comitup[2062]:     return self.ProxyObjectClass(self, bus_name, object_path,
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 250, in __init__
Sep 30 15:00:48 raspberrypi comitup[2062]:     self._named_service = conn.activate_name_owner(bus_name)
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 182, in activate_name_owner
Sep 30 15:00:48 raspberrypi comitup[2062]:     self.start_service_by_name(bus_name)
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 277, in start_service_by_name
Sep 30 15:00:48 raspberrypi comitup[2062]:     return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
Sep 30 15:00:48 raspberrypi comitup[2062]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
Sep 30 15:00:48 raspberrypi comitup[2062]:     reply_message = self.send_message_with_reply_and_block(
Sep 30 15:00:48 raspberrypi comitup[2062]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.NetworkManager was not provided by any .service files
Sep 30 15:00:48 raspberrypi systemd[1]: comitup.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit comitup.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Sep 30 15:00:48 raspberrypi systemd[1]: comitup.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit comitup.service has entered the 'failed' state with result 'exit-code'.
Sep 30 15:00:48 raspberrypi systemd[1]: Failed to start Comitup Wifi Management.
░░ Subject: A start job for unit comitup.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit comitup.service has finished with a failure.
░░ 
░░ The job identifier is 6217 and the job result is failed.`

Any idea? Thank you!

davesteele, 2022-09-30:
The failures are consistent - NetworkManager is not running. That is where to start.

    systemctl status NetworkManager

Perhaps there is an issue with the new 9/26 Pi OS image. I'll try running a new image build

Note. Your installation instructions say to start with Raspberry Pi OS Lite, then install stuff. My image is just that (32 bit), with Comitup installed (and the ssh fix).

outdoorbits, 2022-09-30:
By the way, I tried to install comitup in both, native Raspberry Pi OS lite 32 and 64 bit. Same same.

outdoorbits, 2022-09-30:
Now my original post is better readable. To me this looks very much as a bug inside the python-package.

davesteele, 2022-09-30:
I'm currently having a problem generating images, but I've determined that it is not related to this. There are no significant changes in the last release Pi OS image.

You should concentrate on why the NetworkManager is failing startup.

outdoorbits, 2022-09-30:
It's working again! NetworkManager.service was not enabled.

In my [installation-script](https://github.com/outdoorbits/little-backup-box/blob/main/install-comitup.sh) 

sudo systemctl enable NetworkManager.service
sudo systemctl start NetworkManager.service

directly before the installation of comitup (by apt) did the job.

davesteele, 2022-10-01:
The image build had the same problem. Thanks for the pointer.

closed 2022-10-01 by davesteele