davesteele/comitup #218: Installing comitup when Motioneye has been installed prevents comitup-xxx.local login

evandene, 2022-12-16:
Hi,
I have a usb camera security streaming software installed on my Pi (OS Debian Bullseye). Motioneye is the usb camera streaming software and works great.
When I install comitup, the installation go's smooth without errors. The comitup-xxx is broadcasted and I can SSH into it, so far so good.
What doesn't work is the login via the web browser. The Motioneye  stream opens with the provided http://10.41.0.1/ IP address. The same happens when I try the hostname.local.
Installing comitup without Motioneye installed works great under all circumstances.
I could need some help please

davesteele, 2022-12-17:
The problem is that both Comitup and Motioneye require ownership of port 80. The solution is to allow Comitup to manage the Motioneye service - enabling it when an upstream connection is achieved. See the [Pi-hole](https://davesteele.github.io/raspberrypi/2021/04/29/comitup-and-pihole/) and [Docker](https://github.com/davesteele/comitup/discussions/174) solutions for an idea of how to do this.

evandene, 2022-12-17:
Thanks for the response, the direction given make sense and will get me out of the woods. I will publish the results after the holidays.
Thanks again.

evandene, 2022-12-20:
And indeed all works fine now. No conflicts anymore. Here below the script that worked for me.
First stop Motioneye
sudo systemctl stop motioneye
sudo systemctl disable motioneye
Create a file script with name: /usr/local/bin/comitup-callback 

-	nano /usr/local/bin/comitup-callback
Add the following lines
#!/usr/bin/bash

if [ $1 == "CONNECTED" ] ; then
  systemctl start motioneye
else
  if [ $1 == "HOTSPOT" ] ; then
    systemctl stop motioneye
  fi
fi

Make the script executable

$ sudo chmod 755 /usr/local/bin/comitup-callback
$ sudo chown root.root /usr/local/bin/comitup-callback

Enable motioneye
systemctl enable motioneye

davesteele, 2022-12-23:
> Enable motioneye
> systemctl enable motioneye

You may have an issue after reboot.

closed 2024-07-10 by davesteele