davesteele/comitup #219: Comitup service failed to start

evandene, 2023-01-13:
On my Banana Pi zero with Debian Buster, the Comitup service failed to start.
When looking up the service status i did get the below output.
Could you please help me out with a suggestion?
Can I uninstall comitup? How?
Should I re-install comitup?

pi@clearview:~ $ sudo systemctl status comitup service
Unit service.service could not be found.
● comitup.service - Comitup Wifi Management
   Loaded: loaded (/lib/systemd/system/comitup.service; enabled; vendor preset:
   Active: activating (auto-restart) (Result: exit-code) since Fri 2023-01-13 15
     Docs: man:comitup(8)
  Process: 2063 ExecStart=/usr/sbin/comitup (code=exited, status=1/FAILURE)
 Main PID: 2063 (code=exited, status=1/FAILURE)

Jan 13 15:33:24 clearview systemd[1]: Failed to start Comitup Wifi Management

evandene, 2023-01-13:
I re-installed comitup what did improve but still I have errors. See below the result of systemctl status service:

pi@clearview:~ $ sudo systemctl status comitup service
Unit service.service could not be found.
● comitup.service - Comitup Wifi Management
   Loaded: loaded (/lib/systemd/system/comitup.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2023-01-13 16:09:34 CET; 48s ago
     Docs: man:comitup(8)
 Main PID: 1543 (comitup)
    Tasks: 1 (limit: 854)
   Memory: 17.5M
   CGroup: /system.slice/comitup.service
           └─1543 /usr/bin/python3 /usr/sbin/comitup

Jan 13 16:09:38 clearview dnsmasq[1791]: read /etc/hosts - 5 addresses
Jan 13 16:09:38 clearview dnsmasq[1791]: exiting on receipt of SIGTERM
Jan 13 16:09:41 clearview comitup[1543]: Segmentation fault
Jan 13 16:09:41 clearview comitup[1543]: iptables v1.8.2 (nf_tables): unknown option "--icmp-type"
Jan 13 16:09:41 clearview comitup[1543]: Try `iptables -h' or 'iptables --help' for more information.
Jan 13 16:09:41 clearview comitup[1543]: iptables v1.8.2 (nf_tables): unknown option "--icmp-type"
Jan 13 16:09:41 clearview comitup[1543]: Try `iptables -h' or 'iptables --help' for more information.
Jan 13 16:09:41 clearview comitup[1543]: iptables v1.8.2 (nf_tables): Chain 'COMITUP-OUT' does not exist
Jan 13 16:09:41 clearview comitup[1543]: iptables v1.8.2 (nf_tables): Chain 'COMITUP-OUT' does not exist
Jan 13 16:09:41 clearview comitup[1543]: Try `iptables -h' or 'iptables --help' for more information.

davesteele, 2023-01-13:
I haven't tried a Banana Pi, but this should work.

*iptables* is used internally by Comitup to manage firewall rules between the HOTSPOT and CONNECTED states. The *--icmp-type* option is used to ignore host and port unavailable packets that might be thrown by a captive portal.

I would normally blame the old version of Debian used, but the command in question was added in the Jessie era.

Your problem is that the iptables command does not appear to support the ICMP extension - e.g.:

    iptables -A INPUT -p icmp --icmp-type destination-unreachable -j DROP

evandene, 2023-01-14:
after a full-upgrade, it looks as if the system is in a healthier position now, although I must admit that all is way above my knowledge, so hard to judge by me :-)

pi@clearview:~ $ systemctl status comitup service
Unit service.service could not be found.
● comitup.service - Comitup Wifi Management
   Loaded: loaded (/lib/systemd/system/comitup.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2023-01-14 00:44:35 CET; 24min ago
     Docs: man:comitup(8)
 Main PID: 1556 (comitup)
    Tasks: 2 (limit: 854)
   Memory: 17.4M
   CGroup: /system.slice/comitup.service
           ├─1556 /usr/bin/python3 /usr/sbin/comitup
           └─2457 dnsmasq --conf-file=/usr/share/comitup/dns/dns-hotspot.conf --interface=wlan0

Jan 14 00:45:48 clearview comitup[1556]:     set_device_listeners(modemgr.get_ap_device(), modemgr.get_link_device())
Jan 14 00:45:48 clearview comitup[1556]:   File "/usr/share/comitup/comitup/nmmon.py", line 168, in set_device_listeners
Jan 14 00:45:48 clearview comitup[1556]:     second_device_name != second_dev.Interface
Jan 14 00:45:48 clearview comitup[1556]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 169, in get_func
Jan 14 00:45:48 clearview comitup[1556]:     data = self.proxy.Get(interface, name, dbus_interface='org.freedesktop.DBus.Properties')
Jan 14 00:45:48 clearview comitup[1556]:   File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
Jan 14 00:45:48 clearview comitup[1556]:     **keywords)
Jan 14 00:45:48 clearview comitup[1556]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
Jan 14 00:45:48 clearview comitup[1556]:     message, timeout)
Jan 14 00:45:48 clearview comitup[1556]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop

evandene, 2023-01-14:
Not sure what it means when a moment later the status is different.

pi@clearview:~ $ systemctl status comitup service
Unit service.service could not be found.
● comitup.service - Comitup Wifi Management
   Loaded: loaded (/lib/systemd/system/comitup.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2023-01-14 00:44:35 CET; 27min ago
     Docs: man:comitup(8)
 Main PID: 1556 (comitup)
    Tasks: 2 (limit: 854)
   Memory: 17.3M
   CGroup: /system.slice/comitup.service
           ├─1556 /usr/bin/python3 /usr/sbin/comitup
           └─2457 dnsmasq --conf-file=/usr/share/comitup/dns/dns-hotspot.conf --interface=wlan0

Jan 14 00:45:48 clearview comitup[1556]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
Jan 14 00:45:48 clearview comitup[1556]:     message, timeout)
Jan 14 00:45:48 clearview comitup[1556]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Netwo
Jan 14 01:09:37 clearview comitup[1556]: OK
Jan 14 01:09:37 clearview comitup[1556]: OK
Jan 14 01:09:38 clearview dnsmasq[2457]: reading /etc/resolv.conf
Jan 14 01:09:38 clearview dnsmasq[2457]: using nameserver 8.8.8.8#53
Jan 14 01:10:03 clearview dnsmasq[2457]: reading /etc/resolv.conf
Jan 14 01:10:03 clearview dnsmasq[2457]: using nameserver 195.130.131.1#53
Jan 14 01:10:03 clearview dnsmasq[2457]: using nameserver 195.130.130.1#53

davesteele, 2023-01-14:
Systemd is configured to restart Comitup when it crashes, which it is doing on every startup.

It does not see the dbus interface org.freedesktop, which is pretty fundamental to proper operation.

evandene, 2023-01-16:
Thanks for helping me out.
On the Banana Pi M2 Zero I installed a fresh Armbian Bulsseye OS and installed Comitup as root.
All works fine now without any error

closed 2023-01-16 by evandene