davesteele/comitup #221: Installation problems

outdoorbits, 2023-05-05:
Hello, at first thank you very much as you allow my project [littlebackupbox](https://github.com/outdoorbits/little-backup-box) to connect everywhere.
Now I'm facing a problem on my installation routine:
`wget https://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.deb
sudo dpkg -i --force-all davesteele-comitup-apt-source_latest.deb`

installs in /etc/apt/sources.list.d/davesteele-comitup.list
`deb [signed-by=/usr/share/keyrings/davesteele-archive-keyring.gpg] http://davesteele.github.io/comitup/repo comitup main`
But this link [http://davesteele.github.io/comitup/repo](http://davesteele.github.io/comitup/repo) leads to 404 File not found.

Of course, I could change my installation process, but I would prefer the old way. Do you still maintain https://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.deb?

Thank you!

davesteele, 2023-05-06:
The 404 on the /repo directory is an artifact of GitHub's web service. The repo is there. An "apt update" should verify that.

I verified that the maintenance scripts still update e.g. https://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.deb, and that the link is working for me.

davesteele, 2023-05-06:
A note about your README - for the most part, your users shouldn't need to know about 10.41.0.1. Anything that understands captive portals (Apple stuff, anything running Firefox - I use my smart phone) will load the configuration page automatically.

outdoorbits, 2023-05-06:
Thank you for the note. But using different ports, comitup is also my key to connect my web-UI by this IP but another port number. But I will mention this function in my readme.

outdoorbits, 2023-05-06:
I just tried it again and it's true, `apt update` shows no problems. A user opened an issue and the artefact link looked like the solution to me. Thank you again!

davesteele, 2023-05-06:
> But using different ports, comitup is also my key to connect my web-UI by this IP but another port number

There are a number of strategies for detecting captive portals. The standard one, involving DHCP, can support moving to different ports.

But, you would probably be better served by [sharing the port](https://github.com/davesteele/comitup/discussions/174).

closed 2023-05-07 by davesteele