davesteele/comitup #227: [feature] Import existing networks from wpa_supplicant

Miniontoby, 2023-07-09:
Hi,

Please add a way to (automaticly) import existing wifi network from wpa_supplicant (when installing)

I installed this a month ago on my pi zero which had already internet, but it didn't use the networks that were already in /etc/wpa_supplicant.conf (or how it is called).

If you would add this, then that would be nice. I know like nearly everyone installs this on a new pi, but I didn't and I saw this missed chance.

Kind regards,
- Miniontoby

davesteele, 2023-07-09:
Comitup uses NetworkManager, which obsoletes wpa_supplicant.conf (and the interfaces file).

Comitup will use any predefined NetworkManager WiFi connections.

Miniontoby, 2023-07-10:
> Comitup will use any predefined NetworkManager WiFi connections.

But why did I have to connect to the captive portal it creates to then be able to connect to my wifi network?

I already had internet when installing it, so there was already a wifi network specified.

So I don't understand why it didn't already use it

davesteele, 2023-07-10:
I have no plans to do that, but I endeavor to review patches promptly.

Miniontoby, 2023-07-10:
What command is ran to connect to a new network in this thing? Cause I can find dbus controlled, but I cannot find the code of the comitup dbus

davesteele, 2023-07-10:
comitup-cli

The Comitup class in statemgr.py.

davesteele, 2023-07-10:
... and look at the man pages.

Miniontoby, 2023-07-11:
Here is a script to import from wpa_supplicant:

(it needs the comitup python lib to work, but it could be added in the `apt install` script after the lib has been installed)

```py
#!/usr/bin/env python3

from re import finditer

try:
    from comitup.nm import make_connection_for
except:
    print ("[ERROR] Comitup python lib missing!")
    quit()

try:
    f = open("/etc/wpa_supplicant/wpa_supplicant.conf", "r")
except FileNotFoundError:
    print ("[WARNING] No wpa_supplicant file, skipping import")
    quit()

wpa_conf = f.read()
f.close()

matches = finditer(r"network=\{\n([\s\t]+(ssid=\"([^\"]+)\"|psk=\"([^\"]+)\"|[\w\=\"-_]+)\n)+\}( |)", wpa_conf)
for match in matches:
    ssid = match.group(3)
    psk = match.group(4)
    print ("[INFO] Adding {ssid} {psk}".format(ssid=ssid, psk="without psk" if psk is None else "with psk"))
    make_connection_for(ssid, psk)
```

davesteele, 2023-07-11:
Thanks. It's not quite ready for inclusion in the package, but I've linked it from the [Install Page](https://github.com/davesteele/comitup/wiki/Installing-Comitup). That should be on the path for those who would need it.

closed 2024-07-10 by davesteele