davesteele/comitup #230: 100% CPU Usage in Python 'commit up' Process

Ataraxiall, 2023-09-19:
Hello, I have noticed that the Python process 'commit up' experiences long spikes of 100% CPU usage. Is this normal? I have tried different Python versions, and it happens consistently.

[image]

Thanks!

davesteele, 2023-09-19:
Shouldn't be. That process is for the web page, which only runs in the HOTSPOT mode.

The app does little more than make D-Bus calls from a Flask environment.

Ataraxiall, 2023-09-19:
Well, I've noticed that on different devices with different versions of Python. I don't know what it could be, and I'm not sure if I can see within Python which part of the code is increasing the CPU usage. On the other hand, I've also noticed that on iPhone, the captive portal doesn't show the networks as quickly as on Android. I don't know if there's a connection. 
Thank you and greetings.