davesteele/comitup #231: Comitup-Web no longer starts (after some updates?)

rehflux, 2023-09-23:
I am very sorry, but I suspect I have created a problem for myself. Since I last connected to a wifi using Comitup, I have installed and updated some packages using apt-get and pip. Now the Comitup web service no longer starts and I can only connect to new networks using comitup-cli. I get the following output when I try to open comitup-web in the shell:
``
Traceback (most recent call last):
  File "/usr/sbin/comitup-web", line 5, in <module>
    from web.comitupweb import main
  File "/usr/share/comitup/web/comitupweb.py", line 22, in <module>
    from flask import (
  File "/usr/local/lib/python3.9/dist-packages/flask/__init__.py", line 14, in <module>
    from jinja2 import escape
ImportError: cannot import name 'escape' from 'jinja2' (/usr/local/lib/python3.9/dist-packages/jinja2/__init__.py)
``

I would be very grateful if you could tell me how to fix the problem!
Many thanks in advance!

davesteele, 2023-09-23:
I've not seen this. It sounds like jinja2 needs to be updated.

rehflux, 2023-09-24:
This fixed it:

`pip install --ignore-installed Flask>=2.2.2`
(as root / sudo)

From: https://stackoverflow.com/questions/71718167/importerror-cannot-import-name-escape-from-jinja2

closed 2023-09-27 by davesteele