davesteele/comitup #232: Can't connect to public wifi that contains a captive portal

roshin8, 2023-09-30:
Hi, I'm trying to connect to hotel wifi that has no password but once I connect to it, it requires me to sign in/authenticate. How to handle that scenario with comitup?
Because I connect to the wifi using comitup's captive portal without specifying a password but comitup is still offline because it can't do the next step of signing in/authenticating in the browser

davesteele, 2023-09-30:
If I understand you right, you are connecting to the hotel wifi via the comitup hotspot, but don't get the next step of seeing the hotel login page, because your phone is no longer connected to the comitup device.

The first step to address this is to add a second Wifi interface to the comitup device, and operate in appliance mode. Then there is an opportunity for the phone to see the upstream hotel login page.

I haven't attempted this for a long time. In the old days, the lowest common denominator mechanism for making the login screen visible was to redirect all traffic to the login page. This worked pretty much everywhere, whether or not the client made considerations for a captive portal. Unfortunately, for this to work, the traffic had to be unencrypted. For HTTPS, the cert verification trail would fail. Now that browsers are restricting HTTP, this mechanism is becoming more likely to fail. The other supported method, using DHCP, doesn't map the information beyond the local (upstream) network connection.

In short, add the second interface, and see if that works. Let me know either way.

roshin8, 2023-10-01:
Thank you. How to "operate in appliance mode"?

davesteele, 2023-10-03:
Look at the documentation for *enable_appliance_mode* (and maybe *primary_wifi_device*, if the internally-selected primary wifi device doesn't support AP mode(see the [FAQ](https://github.com/davesteele/comitup/wiki/FAQ)) under "[*man comitup.conf*](https://github.com/davesteele/comitup/blob/main/doc/comitup-conf.5.md)".