davesteele/comitup #233: Simulate internet connection

Ataraxiall, 2023-10-04:
Hello,
 I have a question. I use a hotspot to connect to my devices that don't have internet access so I can work via SSH (I don't use the Raspberry Pi's Wi-Fi connection feature all the time). When I connect, it detects it as an internet-less connection, and it's unstable, disconnecting at times even if I instruct my mobile not to disconnect. Is there a way to make the mobile think that the hotspot has internet? Could it be a DNS server at 10.41.0.1, and any domain resolves to its IP? 
Thanks and regards.

davesteele, 2023-10-04:
I want to say it is a NetworkManager issue, but there is a possible cause from Comitup.

I've encountered a issue with long-lived Access Point sessions. The wpa supplicant process silently crashes, making the AP unusable. This is especially a problem in the HOTSPOT mode, often requiring the device to be rebooted if left in HOTSPOT for a period of time.

I worked around this by periodically killing and restarting the wpa supplicant process. Sometimes, this causes connections to drop, but not always.

The potential problem comes from the fact that I continue the process for the "AP Wifi Device" also while in the connected mode. If you only have one Wifi interface on your device, that could be bad.

Try commenting out [this line](https://github.com/davesteele/comitup/blob/74270364eb93e3a6abf326fc10075e121fb28034/comitup/states.py#L293) from /usr/share/comitup/comitup/states.py, and see if that helps. Let me know.

davesteele, 2023-10-04:
I'll also note that I use the app for the same purpose, though connected to a Wifi AP with an upstream Internet connection, with no problems. That suggests NetworkManager (or a DE?) is the cause.

Take a look at the "CONNECTIVITY SECTION" in the NetworkManager.conf man page.

closed 2024-03-11 by davesteele