davesteele/comitup #234: comitup with newest raspios Bookworm fails

falkyre, 2023-10-12:
I'm trying to use comitup on the just released RaspiOS Bookworm lite (official release October 10) and I am getting the following error:

`Traceback (most recent call last):
  File "/usr/sbin/comitup", line 5, in <module>
    from comitup.comitup import main
  File "/usr/share/comitup/comitup/comitup.py", line 22, in <module>
    from comitup import cdns  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/comitup/comitup/cdns.py", line 14, in <module>
    from comitup import modemgr
  File "/usr/share/comitup/comitup/modemgr.py", line 16, in <module>
    from comitup import config, nm
  File "/usr/share/comitup/comitup/nm.py", line 24, in <module>
    import NetworkManager as nm
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 773, in <module>
    NetworkManager = NetworkManager()
                     ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 251, in __new__
    SignalDispatcher.listen_for_restarts()
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 43, in listen_for_restarts
    dbus.SystemBus().add_signal_receiver(self.handle_restart, 'NameOwnerChanged', 'org.freedesktop.DBus')
    ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/_dbus.py", line 192, in __new__
    return Bus.__new__(cls, Bus.TYPE_SYSTEM, mainloop=mainloop,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/_dbus.py", line 99, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 120, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.FileNotFound: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory

KeyError: dbus.UInt32(32, variant_level=1)
`

Not sure what's changed from Bullseye or how to fix it but I figured I'd let you know.   I'm seeing the same error on one of my plain Debian Bookworm VMs as well.

davesteele, 2023-10-12:
Yup. Debian 12 (Bookworm) is incompatible with the required python-networkmanager, which is [no longer maintained](https://github.com/seveas/python-networkmanager). I'm looking into replacing it.

falkyre, 2023-10-12:
That's what I thought.  I'm hoping it's not a massive rewrite as your app has become a key bit to my image creation.  I'll stick with Bullseye for now and hope you can get things worked out.  I'll leave this open in case anyone else wanders this way for the same reason.

Ataraxiall, 2023-10-13:
> Yup. Debian 12 (Bookworm) is incompatible with the required python-networkmanager, which is [no longer maintained](https://github.com/seveas/python-networkmanager). I'm looking into replacing it.

maybe this can be a option https://github.com/python-sdbus/python-sdbus-networkmanager

davesteele, 2023-10-14:
> I'm hoping it's not a massive rewrite

It is, but I've been interested in reworking it with asyncio.

Pupwiz, 2023-11-12:
any progress? I would like to move some of my portable debian unit to a newer release.
I would like to help but not nearly smart enough to work on this.

davesteele, 2023-11-12:
I'm doing a pretty-much full rewrite in async. It will happen, but is probably a couple months out.

ampledata, 2023-12-29:
I wouldn't discourage your work to add Async, but in the interim I've patched python-networkmanager to fix this Loopback/32 bug:

https://github.com/snstac/python-networkmanager

Pupwiz, 2024-01-07:
Static hostname: zimaboard
Operating System: Debian GNU/Linux 12 (bookworm)  
          Kernel: Linux 6.1.0-17-amd64
    Architecture: x86-64
 Hardware Vendor: IceWhale Technology Co.,Ltd.
  Hardware Model: ZimaBoard 832
Firmware Version: 5.12

Tested and working with patch neworkmanager.
Much appreciated ampledata

Thanks again

davesteele, 2024-01-08:
[python3-networkmanager v2.2-3](https://davesteele.github.io/comitup/deb/python3-networkmanager_2.2-3_all.deb) includes the patch, as well as a general fix for previously unknown device types. It is available in the [Comitup repository](https://davesteele.github.io/comitup/ppa.html).

Leaving this open since it likely won't make it to Pi OS Bookworm.

Pupwiz, 2024-01-08:
Thanks davesteel. noticed during some maintenance on portable server that it did pull the latest deb package and installed. Over writting the patched version without issues.
Thanks again.

Steve

closed 2024-03-11 by davesteele