davesteele/comitup #236: Android No capitive logon comitup 38.1

HenriApperloo, 2023-11-14:
No capitive logon by Android different versions, Apple works fine. I use comitup 38.1.

When I go manual to https://10.41.0.1, then works it good.

Miniontoby, 2023-12-28:
I also have the same problem that the captive portal doesn't show up on Android.

I went ahead and tested a different captive portal software called [nodogsplash](https://github.com/nodogsplash/nodogsplash).

What I did is: I made an AP using hostapd, started nodogsplash and logged in on my Android phone (and windows laptop).
And guess what: It DOES show me that I need to login to the network and it does show me the captive portal login page.

So that confirms that this isn't a bug in Android. 

So the next step would be that someone tries to implement/recreate the code from nodogsplash in comitup. 
And yes I know this might be tricky task since nodogsplash is written in C

Kind regards, 
Miniontoby

davesteele, 2024-03-11:
I've not encountered this with android.

closed 2024-03-11 by davesteele