davesteele/comitup #240: ExecStart=/usr/sbin/comitup (code=exited, status=1/FAILURE)

ZouheirN, 2024-04-03:
Started getting this out of nowhere. Used to work perfectly fine before. Tried updating comitup to latest version, tried rebooting, not working. I am using the "image_2024-01-08-Comitup-lite" iso.

Traceback (most recent call last):
   File "/usr/sbin/comitup", line 8, in <module>
     sys.exit(main())
              ^^^^^^
   File "/usr/share/comitup/comitup/comitup.py", line 123, in main
     (conf, data) = config.load_data()
                    ^^^^^^^^^^^^^^^^^^
   File "/usr/share/comitup/comitup/config.py", line 92, in load_data
     data = persist.persist(
            ^^^^^^^^^^^^^^^^
   File "/usr/share/comitup/comitup/persist.py", line 37, in __init__
     self._load()
   File "/usr/share/comitup/comitup/persist.py", line 47, in _load
     dct = json.load(fp)
           ^^^^^^^^^^^^^
   File "/usr/lib/python3.11/json/__init__.py", line 293, in load
     return loads(fp.read(),
            ^^^^^^^^^^^^^^^^
   File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
     return _default_decoder.decode(s)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
     raise JSONDecodeError("Expecting value", s, err.value) from None
 json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
Apr 03 20:56:09 cropsync-999 systemd[1]: comitup.service: Main process exited, code=exited, status=1/FAILURE

davesteele, 2024-04-03:
It is trying to load persistent data stored in */var/lib/comitup/comitup.json*. 

Is the file present? Can root read it? Is it valid JSON?

I think the only thing stored is the "id" - the number used for the "comitup-&lt;nnn&gt;" name. If the file is corrupt, you can replace it with [this](https://github.com/davesteele/comitup/blob/main/conf/comitup.json), and comitup will re-generate an id.

ZouheirN, 2024-04-03:
Hmm weird, the file was empty. Not sure how that could've happened.

Anyways, I replaced the file with the one you provided, and it works! Thanks for the help.

closed 2024-04-04 by ZouheirN