davesteele/comitup #244: Revert to default on reboot

ryanjAA, 2024-05-30:
I'd like to get this to revert to default on any boot and have done a few things to no avail.

I put a startup script:

```
#!/bin/bash

LOGFILE=/var/log/start_comitup.log

echo "Starting script at $(date)" >> $LOGFILE

# Check if comitup is running and start if necessary
if ! pgrep -x "comitup" > /dev/null
then
    echo "Comitup is not running. Starting comitup..." >> $LOGFILE
    sudo comitup &
else
    echo "Comitup is already running." >> $LOGFILE
fi

# Wait for comitup to start
for i in {1..30}; do
    if pgrep -x "comitup" > /dev/null
    then
        echo "Comitup started successfully." >> $LOGFILE
        break
    else
        echo "Waiting for comitup to start... ($i)" >> $LOGFILE
        sleep 2
    fi
done

# Run comitup-cli d
comitup-cli d >> $LOGFILE 2>&1
```

that runs on boot but that strangely doesnt do it

I also changed the cli.py do_delete function to:

```
def do_delete(ciu_client, connection):
    # List of names to exclude
    excluded_names = ["comitup", "dhcp", "lo", "static"]

    # Check if the connection name starts with any of the excluded names
    if any(connection.startswith(name) for name in excluded_names):
        print(f"Skipping deletion of connection '{connection}'")
    else:
        ciu_client.ciu_delete(connection)
 ```       
       
 so you cant delete core functions by accident (possibly could amend so anything in the conf feeds in here instead of comitup but this should maybe go upstream @davesteele (will do a pr if you want) but i found myself deleting network manager configs accidentally without the above, probably safer for all to have that.
 
 Anyway - @davesteele what am i missing to get this to revert to new (aka delete whatever wifi i've joined on the prev( and then on reboot? I messed with nuke but rpi.gpio doesnt seem to work (anymore?) and I quasi got lgpio but abandoned due to not thinking that was the correct path...
 
 Much appreciated Dave - this is great!

ryanjAA, 2024-05-30:
Ok, got it working but dont love how:

added /etc/systemd/system/comitup.service
 ```
 [Unit]
Description=Comitup Service
After=network.target

[Service]
ExecStart=/usr/sbin/comitup
Restart=on-failure
RestartSec=10

[Install]
WantedBy=multi-user.target
```
sudo systemctl daemon-reload
sudo systemctl enable comitup
sudo systemctl start comitup
 
added: /usr/local/bin/start_comitup.sh
```
#!/bin/bash

LOGFILE=/var/log/start_comitup.log

echo "Starting script at $(date)" >> $LOGFILE

# Wait for comitup to start
for i in {1..30}; do
    if systemctl is-active --quiet comitup
    then
        echo "Comitup started successfully." >> $LOGFILE
        break
    else
        echo "Waiting for comitup to start... ($i)" >> $LOGFILE
        sleep 2
    fi
done

# Double-check if comitup is still running
if systemctl is-active --quiet comitup
then
    echo "Running comitup-cli d" >> $LOGFILE
    comitup-cli d >> $LOGFILE 2>&1
else
    echo "Comitup did not start successfully." >> $LOGFILE
fi
```
sudo chmod +x /usr/local/bin/start_comitup.sh (aka make it executable)
added: /etc/systemd/system/comitup_start.service
```
[Unit]
Description=Run Comitup CLI on boot after Comitup starts
After=comitup.service

[Service]
ExecStart=/usr/local/bin/start_comitup.sh
Restart=on-failure
RestartSec=10
User=root
StandardOutput=journal
StandardError=journal

[Install]
WantedBy=multi-user.target
```
sudo systemctl daemon-reload
sudo systemctl enable comitup_start
sudo systemctl start comitup_start

sudo reboot

and works

davesteele, 2024-05-30:
Personally, I would write a startup script that uses nmcli to identify and delete unwanted connections (it might be easier to write a python script that acesses python-networkmanager to do that).

There's no need to manage comitup to the degree you are doing here. It should adapt to the deleted connections.

ryanjAA, 2024-05-30:
I like it. Thanks. I'll give it a quick try. 

So if deleting all connections but ones that exist on a clean install boot, the acid test will be if comitup goes back to broadcasting the standard AP said without another reboot (I could add another reboot in but that would add some latency).

davesteele, 2024-05-30:
You should not need an extra reboot. It should adapt to the deleted connection.

ryanjAA, 2024-05-30:
# reset_networks.py

import NetworkManager

def delete_unwanted_connections():
    unwanted_connections = []
    for conn in NetworkManager.Settings.ListConnections():
        settings = conn.GetSettings()
        id = settings['connection']['id']
        if not id.startswith(('comitup', 'dhcp', 'static', 'lo')):
            unwanted_connections.append(id)

    for conn_id in unwanted_connections:
        conn = NetworkManager.Settings.Connection.GetSettings(conn_id)
        conn.Delete()
        print(f"Deleted connection: {conn_id}")

if __name__ == "__main__":
    delete_unwanted_connections()

ryanjAA, 2024-05-30:
> Make sure NetworkManager has a chance to come up first.

Oh good point. 👍👍👍

Still like this in cli to not kill everything by accident (which I did a bunch of times 🙄🙄🙄)

closed 2024-07-10 by davesteele