davesteele/comitup #245: Errors when installing via apt (RaspberryPi OS: Bookworm)

ItsVRK, 2024-06-15:
```
Setting up comitup (1.41-1) ...
Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.
comitup-web.service is a disabled or a static unit not running, not starting it.
Processing triggers for dbus (1.14.10-1~deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...
```

ItsVRK, 2024-06-18:
Additional information:

```
journalctl -xeu comitup.service
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/share/comitup/comitup/nmmon.py", line 190, in init_nmmon
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     set_device_listeners(modemgr.get_ap_device(), modemgr.get_link_device())
Jun 19 09:50:28 plexamp-lounge comitup[1485]:                          ^^^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/share/comitup/comitup/modemgr.py", line 50, in get_ap_device
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     devs = nm.get_wifi_devices()
Jun 19 09:50:28 plexamp-lounge comitup[1485]:            ^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/share/comitup/comitup/nm.py", line 81, in get_wifi_devices
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     devices: Optional[List[nm.Device]] = get_devices()
Jun 19 09:50:28 plexamp-lounge comitup[1485]:                                          ^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/share/comitup/comitup/nm.py", line 65, in get_devices
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     device_list = nm.NetworkManager.GetDevices()
Jun 19 09:50:28 plexamp-lounge comitup[1485]:                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "<string>", line 8, in GetDevices
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 583, in to_python
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     val = fixups.base_to_python(val)
Jun 19 09:50:28 plexamp-lounge comitup[1485]:           ^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 646, in base_to_python
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     return [fixups.base_to_python(x) for x in val]
Jun 19 09:50:28 plexamp-lounge comitup[1485]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 646, in <listcomp>
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     return [fixups.base_to_python(x) for x in val]
Jun 19 09:50:28 plexamp-lounge comitup[1485]:             ^^^^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 659, in base_to_python
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     return globals()[classname](val)
Jun 19 09:50:28 plexamp-lounge comitup[1485]:            ^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 353, in __new__
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     klass = device_class(obj.Get('org.freedesktop.NetworkManager.Device', 'DeviceType', dbus_interface='org.freedesktop.DBus.Properties'))
Jun 19 09:50:28 plexamp-lounge comitup[1485]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 19 09:50:28 plexamp-lounge comitup[1485]:   File "/usr/lib/python3/dist-packages/NetworkManager.py", line 373, in device_class
Jun 19 09:50:28 plexamp-lounge comitup[1485]:     return {
Jun 19 09:50:28 plexamp-lounge comitup[1485]:            ^
Jun 19 09:50:28 plexamp-lounge comitup[1485]: KeyError: dbus.UInt32(32, variant_level=1)
Jun 19 09:50:28 plexamp-lounge systemd[1]: comitup.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit comitup.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Jun 19 09:50:28 plexamp-lounge systemd[1]: comitup.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit comitup.service has entered the 'failed' state with result 'exit-code'.
Jun 19 09:50:28 plexamp-lounge systemd[1]: Failed to start comitup.service - Comitup Wifi Management.
░░ Subject: A start job for unit comitup.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit comitup.service has finished with a failure.
░░
░░ The job identifier is 3821 and the job result is failed.
lines 1740-1793/1793 (END)
```

ItsVRK, 2024-06-19:
Was able to successfully install on a clean bookworm install by following the "Better Way" instructions at https://davesteele.github.io/comitup/ppa.html

closed 2024-06-19 by ItsVRK

davesteele, 2024-06-19:
Thanks for the report.

For what you are doing, the "Better Way" didn't provide anything extra to fix the problem - it just installed the (latest) version you were already working with. The clean install must have been the secret. That was an odd failure.

This issue prompted me to do a cleaner job of installing the systemd services.