davesteele/comitup #247: Rename Wifi SSID

rajavel1990, 2024-06-20:
I am trying to rename the Wifi SSID from comitup-586 to the Project name that we are building like "ABC Techtronics" how do we get this changed to reflect the project name as the Wifi SSID ?

davesteele, 2024-06-20:
See the (ap_name)[https://davesteele.github.io/comitup/man/comitup-conf.pdf] configuration parameter.

rajavel1990, 2024-06-20:
I have tried changing the ap_name but i dont see the changes it is still displaying comitup-586.

[image]
[image]

davesteele, 2024-06-20:
The "#" is a comment character. Remove it and the leading space.

rajavel1990, 2024-06-21:
This Worked. Thanks David.,

closed 2024-06-21 by davesteele