davesteele/comitup #248: Need to Configure Comitup in Existing project

rajavel1990, 2024-06-21:
I have followed the instructions on this page: https://github.com/davesteele/comitup/wiki/Installing-Comitup to get comitup configured. 

I have successfully executed the first command sudo apt-get install comitup

When i run the 2nd command sudo systemctl enable NetworkManager.service i am getting a error Failed to enable unit: Unit file NetworkManager.service does not exist.
 how do i get past this to proceed with the next steps ? 

Is there any video tutorial for manual installation on preinstalled raspbian os ?

[image]

davesteele, 2024-06-21:
Installing comitup should have pulled in NetworkManager, by way of python-networkmanager. 

Is it installed?

    dpkg --list | grep network-manager

If it is, try rebooting to reset systemd.

rajavel1990, 2024-06-21:
I just figured out that i am running Buster and there were some additional steps that needs to be done. Is there a comitup image that has buster installed. The one that I download is with the latest bookworm

rajavel1990, 2024-06-22:
I have followed all the instructions provided in https://github.com/davesteele/comitup/wiki/Installing-Comitup and https://davesteele.github.io/comitup/ppa.html to get the comitup installed without any errors., but after i reboot i dont see any wifi displayed i connected an ethernet cable to ssh into the pi. I logged in through VNC and i see the the wifi is disconnected and it displays 

No wireless interfaces found
connection to dhcpcd lost

What should be done now to fix this ?

davesteele, 2024-06-22:
Network manager needs to own the WiFi interfaces. That means removing/commenting them in /etc/interfaces.

rajavel1990, 2024-06-23:
I dont see any file by the name in etc/interfaces but i have deleted the file named interfaces in etc/network/interfaces

[image]

closed 2024-07-03 by davesteele