davesteele/comitup #249: Possible to speed up shutdown time?

BrianGilbert, 2024-06-25:
commitup is running a process that slows down shutdown, it takes 1.5 minutes.. any way to reduce that timer?

davesteele, 2024-06-26:
Probably. It should shut down instantaneously. I haven't looked into this.

davesteele, 2024-06-26:
Can't reproduce.

BrianGilbert, 2024-06-26:
@davesteele apologies, it was reboot time that has the pause..

using `sudo shutdown -r now` I get the following during the shutdown procedure

[image]

davesteele, 2024-06-27:
Take a look in the comitup (/var/log/comitup.log) and systemd (journalctl -u comitup) logs.

davesteele, 2024-06-27:
Comitup is having trouble with NetworkManager. Can you try adding "KillSignal=SIGKILL" to the [Service] section of the comitup.service file (and run "systemctl daemon-reload")?

closed 2024-07-03 by davesteele