davesteele/comitup #250: Comitup DNS server

Ataraxiall, 2024-07-06:
Good morning,

I want to make it so that when a client is connected, all DNS requests are redirected to the localhost of comitup. This is because the Raspberry Pi does not have internet, and the iPhone disconnects from the access point after a few minutes due to the lack of internet. I want to try installing Nginx so that the requests checking for internet connectivity are resolved locally.

Here is the explanation for the Apple website configuration:

- **First `server` block:**
  ```nginx
  server {
      listen 80;
      server_name connectivitycheck.gstatic.com;

      location /generate_204 {
          return 204;
      }
  }
  ```

  - Listens on port 80 for the domain `connectivitycheck.gstatic.com`.
  - Returns a 204 status code for the `/generate_204` path (Android connectivity check).

- **iPhone connectivity check:**
  The iPhone checks for internet connectivity by trying to reach `http://www.apple.com/library/test/success.html`. It expects a response containing the text "Success". If it receives this response, it assumes that there is an internet connection.

Is it possible to do this? In which file do I put the DNS resolutions? 
Thank you and best regards.

Ataraxiall, 2024-07-08:
Thanks for the response.

I cant locate dnsmasq conf file (not in /etc).

Where i cand find it?

Thanks!

davesteele, 2024-07-08:
It's owned by Comitup

    dpkg --listfiles comitup | grep dns-

closed 2024-07-10 by davesteele