davesteele/comitup #253: 1.42 fails to start on RaspiOS

mstormi, 2024-08-24:
Running on RaspiOS dated July, 4th, 2024, downloaded from Raspi foundation
I've installed latest comitup package I found in the repo (1.42-1)

Please don't hesitate to ask if you need more information.

Problem is, startup fails (via systemd, but also if directly started like this:)

```
root@openhabian:/home/openhabian# python /usr/sbin/comitup

Traceback (most recent call last):
  File "/usr/sbin/comitup", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/share/comitup/comitup/comitup.py", line 149, in main
    statemgr.init_state_mgr(
  File "/usr/share/comitup/comitup/statemgr.py", line 200, in init_state_mgr
    states.init_states(
  File "/usr/share/comitup/comitup/states.py", line 430, in init_states
    nmmon.init_nmmon()
  File "/usr/share/comitup/comitup/nmmon.py", line 190, in init_nmmon
    set_device_listeners(modemgr.get_ap_device(), modemgr.get_link_device())
                         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/comitup/comitup/modemgr.py", line 50, in get_ap_device
    devs = nm.get_wifi_devices()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/comitup/comitup/nm.py", line 81, in get_wifi_devices
    devices: Optional[List[nm.Device]] = get_devices()
                                         ^^^^^^^^^^^^^
  File "/usr/share/comitup/comitup/nm.py", line 65, in get_devices
    device_list = nm.NetworkManager.GetDevices()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 8, in GetDevices
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 583, in to_python
    val = fixups.base_to_python(val)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 646, in base_to_python
    return [fixups.base_to_python(x) for x in val]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 646, in <listcomp>
    return [fixups.base_to_python(x) for x in val]
            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 659, in base_to_python
    return globals()[classname](val)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 353, in __new__
    klass = device_class(obj.Get('org.freedesktop.NetworkManager.Device', 'DeviceType', dbus_interface='org.freedesktop.DBus.Properties'))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/NetworkManager.py", line 373, in device_class
    return {
           ^
KeyError: dbus.UInt32(32, variant_level=1)
```

I believe prerequisites are all there:

```
root@openhabian:/home/openhabian# python --version
Python 3.11.2
root@openhabian:/home/openhabian# dpkg -l network-manager
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name            Version       Architecture Description
+++-===============-=============-============-======================================================>
ii  network-manager 1.42.4-1+rpt1 arm64        network management framework (daemon and userspace too>
root@openhabian:/home/openhabian# dpkg -l |grep -i networkmanager
ii  libnm0:arm64                                     1.42.4-1+rpt1                             arm64        GObject-based client library for NetworkManager
ii  python3-networkmanager                           2.2-2                                     all          Python 3 interface to the NetworkManager D-Bus interface
```

davesteele, 2024-08-24:
You need python-networkmanager v2.2-3, available in the [comitup repo](http://davesteele.github.io/comitup/ppa.html) ([changelog](https://tracker.debian.org/media/packages/p/python-networkmanager/changelog-2.2-3)).

mstormi, 2024-08-24:
got it, works now, thanks a lot for your prompt service!

On a sidenote, this showed up in the first because we've been using a bad repo.
We're installing your comitup repo config from within a script of ours (i.e. not interactively), using the
"better way" as you told in http://davesteele.github.io/comitup/ppa.html
The "latest davesteele-comitup-apt-source deb file" however links to another HTML page(http://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.html) and to not the .deb-file itself that we want to fetch to install it with dpkg.
So we would need to parse the HTML to get the ultimate link to the deb.
Please could you ease things and make it a direct link? Or provide some additional direct link to the latest .deb?

Thanks a lot again for all your work on this!

closed 2024-08-24 by mstormi

davesteele, 2024-08-24:
> Please could you ease things and make it a direct link?

Done