davesteele/comitup #258: Apache2 and Comitup

dubaleeiro, 2024-10-13:
Hi,

This is more a question than an issue. I have been trying to make a Django application working with Apache2 and Comitup on a Raspberry Pi and looked at some issues like [this one](https://github.com/davesteele/comitup/issues/201).

My approach to make Apache2 and Comitup coexist is to start the Apache server only after Comitup successfully connects to a WiFi network by changing the `start-apache.service` like this:

`
[Unit]
Description=Start Apache after WiFi connection
After=comitup.service
`

This did not work unfortunately. Would someone be so kind and give me some hints? Thanks !

davesteele, 2024-10-14:
Just because Comitup is running doesn't mean that WiFi is connected. 

See #142 and the [web_service](https://github.com/davesteele/comitup/blob/8ed8f908b1bfe25ee139fb7d35846e7cb60f0afb/conf/comitup.conf#L28) [configuration parameter](https://davesteele.github.io/comitup/man/comitup-conf.pdf).

dubaleeiro, 2024-10-15:
Changing the `web_service` parameter to `apache2.service` did the trick !

Thanks a lot, @davesteele !

closed 2024-10-15 by davesteele