davesteele/comitup #260: Page for Wifi connections not working

dondidonderidon, 2024-12-22:
Hi,

I followed the instructions and was able to get comitup working on top of an existing OS.
I copied the microsd card and found that the wifi connection is not working on the copy. 
The comitup hotspot has the same name as the copy but when I try to connect to it, it doenst load the page or gives unreachable.
Using the original microsd card works fine. Is there something I did wrong for it not to work like this?

Thank you

davesteele, 2024-12-24:
Check the logs in /var/log.

Did you copy the card after first booting it? There is a partition resize in the first boot that could lead to a corrupted partition on the copy.

Does the card actually boot? Check the logs or hook up a monitor and keyboard.

The most likely culprits have nothing to do with Comitup.

dondidonderidon, 2024-12-25:
> Check the logs in /var/log.
>
> Did you copy the card after first booting it? There is a partition resize in the first boot that could lead to a corrupted partition on the copy.
>
> Does the card actually [...]

I did boot it first to try and the copied yes, I'll try again without booting it first. 
The card does boot as the comitup hotspot shows and after adding my wifi manually it starts as expected.
It just doenst want to connect to wifi via the comitup script. 

Thank you for the help, will try again!

davesteele, 2026-01-14:
Closing. If you see this problem, consult the [wiki](https://github.com/davesteele/comitup/wiki/Installing-Comitup).

closed 2026-01-14 by davesteele