davesteele/comitup #261: Accessing internet through comitup

wspi, 2025-01-31:
I usually use comitup on my raspberry pi 3 connected via ethernet cable, doing so I was wondering, is there a way to comitup detect that the wired connection is on, and turn the wifi on so users could connect to this wifi and have access to the internet?

I've connected to the hotspot ssid but it won't have access to the internet

not sure if #168 would allow this

davesteele, 2025-01-31:
#168 is different. It suggests that Comitup should drop the hotspot if there is a wired connection.

You need two things to be able to do this - ip_forwarding needs to be enabled, and masquerading needs to be set for the wired port. Comitup does both of these things. See the rules with nft when all interfaces are up.

I replaced iptables with nft last year. I may have messed the [masquerading](https://github.com/davesteele/comitup/blob/ea16776c7ff57471ff2830bfd2ae87dda26fd807/comitup/nftmgr.py#L50) up. Let me know if you see a problem with the rules.

wspi, 2025-01-31:
I'm not familiar with nft, but I'll take a look and let you know

wspi, 2025-01-31:
one thing that I've noticed is that ip_forwarding is not enabled, 

```
# cat /proc/sys/net/ipv4/ip_forward
0
```

davesteele, 2025-01-31:
Marking as a bug.

wspi, 2025-02-03:
> Marking as a bug.

thanks, if you need help, just let me know

wspi, 2025-05-15:
I finally had some time and learned enough nft to be able to create masquerading rules

I enable the port forwarding, then created a nat table for masquerading
```
sudo nft add table ip nat
sudo nft add chain ip nat postrouting '{ type nat hook postrouting priority 100; }'
sudo nft add rule ip nat postrouting oifname "eth0" ip saddr 10.41.0.0/24 masquerade
```

I had to change the DNS server on my mobile also, otherwise it would always go to the captive portal... it's just not fast, around 20mb/s but it might be related to hardware (raspberry pi 5)?

davesteele, 2025-05-15:
Changing to enhancement.