davesteele/comitup #265: Switch to captive portal without discarding saved WiFi passwords

m-sofer, 2025-05-01:
Is there a simple way to switch to the captive portal mode without discarding the saved WiFi connection and go back to WiFi mode when needed, I rather not add a second WiFi module. I am adding a GPO button to start a script for going back to the portal mode. To get back to WiFi mode I will be customizing the portal screen and adding a button to go back along with other information I need.

Thanks in advance

davesteele, 2025-05-01:
Not currently.

m-sofer, 2025-05-01:
Can I access the portal page using the Pi's IP address.

davesteele, 2025-05-02:
You could access comitup-cli(1) via ssh.

m-sofer, 2025-05-02:
I need to show status updated to end uses so ssh is not viable. If I run my own python web page as a service through flask, is there a way to use port 80 without breaking the portal, it will be accessed only while the WiFi is connected . As an alternative, can I change the port you are using and if so, which file should I update.

Thank you and I do appreciate your time.

davesteele, 2025-05-03:
See web_service in comitup.conf.

m-sofer, 2025-05-05:
Thank you for creating this captive portal