davesteele/comitup #266: deb repo gpg key

8de2fdb0, 2025-05-02:
Running 

```bash
# curl -LO https://raw.githubusercontent.com/davesteele/comitup/refs/heads/gh-pages/repo/dists/comitup/Release.gpg
# gpg --list-packets < Release.gpg
:signature packet: algo 1, keyid 0959C4A3DCF89FBF
	version 4, created 1730054914, md5len 0, sigclass 0x00
	digest algo 10, begin of digest c2 e0
	hashed subpkt 33 len 21 (issuer fpr v4 8B8490B14A006D36311D2DC50959C4A3DCF89FBF)
	hashed subpkt 2 len 4 (sig created 2024-10-27)
	subpkt 16 len 8 (issuer key ID 0959C4A3DCF89FBF)
	data: [4096 bits]
```

Shows a signature from 8B8490B14A006D36311D2DC50959C4A3DCF89FBF

But 
```
# curl -O https://davesteele.github.io/key-366150CE.pub.txt
# gpg --show-keys key-366150CE.pub.txt
pub   rsa4096 2010-08-15 [SC]
      AE0DBF5A92A5ADE49481BA6F8A3171EF366150CE
uid                      David Steele <steele@debian.org>
uid           [ revoked] David Steele <daves@users.sourceforge.net>
uid                      David Steele <dsteele@gmail.com>
sub   rsa4096 2010-08-15 [E]
sub   rsa4096 2014-08-15 [S] [expired: 2019-08-14]
sub   rsa4096 2017-06-25 [S] [expired: 2022-06-24]
sub   rsa4096 2021-02-23 [S] [expires: 2026-02-22]
```

Shows a different key, same for 
https://github.com/davesteele/davesteele.github.io/blob/master/key-366150CE.pub.txt

Any Idea whats going on?

davesteele, 2025-05-02:
The key file you are inspecting is actually a key certificate, containing a master key, "subkeys", and signatures tying everything together. The key 366150CE (I'm taking the last 8 characters, for the "short id") is the master key, DCF89FBF is one of the subkeys. It's actually represented in the last line of your --show-keys output:

    sub   rsa4096 2021-02-23 [S] [expires: 2026-02-22]

Try this to get more detail on the certificate:

    cat key-366150CE.pub.txt | gpg --list-packets

The key in question is identified in the second-to-last stanza.

If you want to understand this better, read [Anatomy of a GPG Key](https://davesteele.github.io/gpg/2014/09/20/anatomy-of-a-gpg-key/).

8de2fdb0, 2025-05-03:
I had an issue setting up an APT entry using the 

```
deb [signed-by=<keyring file>] http://davesteele.github.io/comitup/repo comitup main
```
The error came from apt not finding the correct public key. So I investigated and I guess I got inspecting the key file wrong.

In any case following entry is now working for me.

```
deb [signed-by=/usr/share/keyrings/davesteele-archive-keyring.gpg] http://davesteele.github.io/comitup/repo comitup main
```

Thanks for the help.

Maybe you should update your repo instructions since apt is now complaining when using apt-key to import the public key.
I think you are discourring people from using the repo, but at least on bookworm it is the easiest way since `python3-networkmanager` isn't available in the version required to work.

closed 2025-05-05 by 8de2fdb0