davesteele/comitup #267: comitup networm manager error on rpi4b with OS (bookworm) Lite

hlima, 2025-05-20:
installed comitup on raspberry pi 4b running OS Lite Bookworm using with the command:
    ` sudo apt install comitup` i
configure according to the instructions on the website.
running the commitup command from a console is giving this error:
`sudo comitup`
`Traceback (most recent call last):`
 ` File "/usr/sbin/comitup", line 8, in <module>`
  `  sys.exit(main())`
             ^^^^^^
 ` File "/usr/share/comitup/comitup/comitup.py", line 149, in main`
  `  statemgr.init_state_mgr(`
 ` File "/usr/share/comitup/comitup/statemgr.py", line 200, in init_state_mgr
    states.init_states(`
  `File "/usr/share/comitup/comitup/states.py", line 430, in init_states`
    `nmmon.init_nmmon()`
 ` File "/usr/share/comitup/comitup/nmmon.py", line 190, in init_nmmon`
   ` set_device_listeners(modemgr.get_ap_device(), modemgr.get_link_device())`
                         ^^^^^^^^^^^^^^^^^^^^^^^
  `File "/usr/share/comitup/comitup/modemgr.py", line 50, in get_ap_device`
    `devs = nm.get_wifi_devices()`
    `       ^^^^^^^^^^^^^^^^^^^^^`
  `File "/usr/share/comitup/comitup/nm.py", line 81, in get_wifi_devices`
    `devices: Optional[List[nm.Device]] = get_devices()`
    `                                     ^^^^^^^^^^^^^`
  `File "/usr/share/comitup/comitup/nm.py", line 65, in get_devices`
   ` device_list = nm.NetworkManager.GetDevices()`
   `               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^`
  `File "<string>", line 8, in GetDevices`
  `File "/usr/lib/python3/dist-packages/NetworkManager.py", line 583, in to_python`
   ` val = fixups.base_to_python(val)`
    `      ^^^^^^^^^^^^^^^^^^^^^^^^^^`
  `File "/usr/lib/python3/dist-packages/NetworkManager.py", line 646, in base_to_python`
    `return [fixups.base_to_python(x) for x in val]`
    `       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^`
 ` File "/usr/lib/python3/dist-packages/NetworkManager.py", line 646, in <listcomp>`
    `return [fixups.base_to_python(x) for x in val]`
   `         ^^^^^^^^^^^^^^^^^^^^^^^^`
 ` File "/usr/lib/python3/dist-packages/NetworkManager.py", line 659, in base_to_python`
  `  return globals()[classname](val)`
   `        ^^^^^^^^^^^^^^^^^^^^^^^^^`
  `File "/usr/lib/python3/dist-packages/NetworkManager.py", line 353, in __new__`
   ` klass = device_class(obj.Get('org.freedesktop.NetworkManager.Device', 'DeviceType', dbus_interface='org.freedesktop.DBus.Properties'))`
            `^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^`
  `File "/usr/lib/python3/dist-packages/NetworkManager.py", line 373, in device_class`
   ` return {`
   `        ^`
`KeyError: dbus.UInt32(32, variant_level=1)`

davesteele, 2025-05-21:
Did you install the patched python3-networkmanager, as mentioned on the [home page](https://davesteele.github.io/comitup/)? 

[Instructions](https://davesteele.github.io/comitup/ppa.html)

hlima, 2025-05-21:
I think I dont have the patched version of net manager. I did not install it directly. I added the davsteele repo, ran apt update an run apt install comitup. I will read the instruction again and see how I can install the patched version.

davesteele, 2025-05-21:
That should have happened with the repo install. What version of python3-networkmanager do you have installed?